atopile/faebryk #40: Library: Improve the elegancy of decoupling power

iopapamanoglou, 2024-09-06:
### Feature Request

- Reference designs often suggest a value, but Constant values are bad
- The whole notion of decoupling is very solution-space instead of problem-space
- Who is responsible for decoupling, how do you decide something needs it

Idea: Move into problem-space
Let modules describe their requirements for power noise & voltage stability instead.
This makes reference designs also more lean

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ruben-iteng, 2024-09-06:
A problem why moving decoupling into problem-space is difficult is because most real components are just black-boxes with the only info available is in the datasheet. We don't know the (exact) power requirements of the component.

The only thing we do know for proper working of the component is that it needs a decoupling cap as described in the datasheet.

I do agree that we need to change the way decoupling works. It should be a more descriptive process and not just defining a static capacitor.

ruben-iteng, 2024-09-06:
Some basic info:

https://www.analog.com/media/en/training-seminars/tutorials/MT-101.pdf

https://en.wikipedia.org/wiki/Decoupling_capacitor

https://www.ti.com/lit/an/spra906/spra906.pdf

mawildoer, 2024-09-10:
Voltage stability and power noise are too abstract in this case.

The most abstract solution here is a capacitance with some "stiffness" on the supply (ESR/ESL). Pragmatically, though, we'd need to back-calculate these from typical decoupling cap designs, and it'd all be very fuzzy anyway.

Decoupling via "a capacitor" is a good solution, as long as which capacitor is selected is abstract.