atopile/faebryk #118: Bug: pytest: error: unrecognized arguments

ruben-iteng, 2024-10-31:
### Current Behaviour

Running `pytest` will give the following error:
``` bash
pytest: error: unrecognized arguments: --numprocesses=auto --dist=loadscope
```

### Expected Behaviour

Just runs the tests

### Possible Solution

Solution (for now):
1. exit your venv by typing: `deactivate` in your terminal
2. re-activate your venv

### Version

4.2.1

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mawildoer, 2024-11-06:
Seems like a venv issue off the cuff. Got anything to indicate that it's something up in the code?

ruben-iteng, 2024-11-07:
Indeed, I posted it here so it's documented. We need to see how serious it is (especially with new installations/users).