atopile/faebryk #121: Iterative Design Example Failing

mawildoer, 2024-11-02:
### Current Behaviour

Fails with KiCAD 7 library error:
[iterative_design_nand.log](https://github.com/user-attachments/files/17607874/iterative_design_nand.log)

System: OSx, Python3.12, KiCAD 8, no KiCAD 7 install
Githash: `df1e405`

Grep results from directory content as well:
```
(.venv) faebryk-wt1 % grep . "${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod"
grep: /Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod: No such file or directory
(.venv) faebryk-wt1 % grep -r . '${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod'
grep: ${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod: No such file or directory
(.venv) faebryk-wt1 % grep -R '${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod' .
./iterative_design_nand.log:│ │                path = PosixPath('${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod')                                                                                                                  │                │
./iterative_design_nand.log:│   12                                                                                          │ path = PosixPath('${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod') │                                                  │
./iterative_design_nand.log:│ ❱ 1027 │   │   with self.open(mode='r', encoding=encoding, errors=errors) as f:                 │     self = PosixPath('${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod') │                                            │
./iterative_design_nand.log:│   1016 │   │   """                                                                              │      self = PosixPath('${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod') │                                           │
./iterative_design_nand.log:FileNotFoundError: [Errno 2] No such file or directory: '${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod'
(.venv) faebryk-wt1 % grep -R 'KICAD7_FOOTPRINT_DIR' .                                             
./iterative_design_nand.log:│ │            dir_path = PosixPath('${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty')                                                                                                                                           │                │
./iterative_design_nand.log:│ │                       │   │   │   C_lib(name='Audio_Module', type='Kicad', uri='${KICAD7_FOOTPRINT_DIR}/Audio_Module.pretty', options='', descr='Audio Module footprints'),                                             │                │
./iterative_design_nand.log:│ │                       │   │   │   C_lib(name='Battery', type='KiCad', uri='${KICAD7_FOOTPRINT_DIR}/Battery.pretty', options='', descr='Battery and battery holder footprints'),                                         │                │
./iterative_design_nand.log:│ │                       │   │   │   C_lib(name='Button_Switch_Keyboard', type='KiCad', uri='${KICAD7_FOOTPRINT_DIR}/Button_Switch_Keyboard.pretty', options='', descr='Buttons and switches for keyboard applications'),  │                │
./iterative_design_nand.log:│ │                       │   │   │   C_lib(name='Button_Switch_SMD', type='KiCad', uri='${KICAD7_FOOTPRINT_DIR}/Button_Switch_SMD.pretty', options='', descr='Buttons and switches, surface mount'),                       │                │
./iterative_design_nand.log:│ │                       │   │   │   C_lib(name='Button_Switch_THT', type='KiCad', uri='${KICAD7_FOOTPRINT_DIR}/Button_Switch_THT.pretty', options='', descr='Buttons and switches, through hole'),                        │                │
./iterative_design_nand.log:│ │                       │   │   │   C_lib(name='Buzzer_Beeper', type='KiCad', uri='${KICAD7_FOOTPRINT_DIR}/Buzzer_Beeper.pretty', options='', descr='Audio signalling devices'),                                          │                │
./iterative_design_nand.log:│ │                       │   │   │   │   uri='${KICAD7_FOOTPRINT_DIR}/Calibration_Scale.pretty',                                                                                                                           │                │
./iterative_design_nand.log:│ │                       │   │   │   C_lib(name='Capacitor_SMD', type='KiCad', uri='${KICAD7_FOOTPRINT_DIR}/Capacitor_SMD.pretty', options='', descr='Capacitor, surface mount'),                                          │                │
./iterative_design_nand.log:│ │                       │   │   │   C_lib(name='Capacitor_THT', type='KiCad', uri='${KICAD7_FOOTPRINT_DIR}/Capacitor_THT.pretty', options='', descr='Capacitor, through hole'),                                           │                │
./iterative_design_nand.log:│ │                       │   │   │   C_lib(name='Capacitor_Tantalum_SMD', type='KiCad', uri='${KICAD7_FOOTPRINT_DIR}/Capacitor_Tantalum_SMD.pretty', options='', descr='Tantalum Capacitor, surface mount'),               │                │
./iterative_design_nand.log:│ │                 lib = C_lib(name='Package_DIP', type='KiCad', uri='${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty', options='', descr='Through hole IC packages, DIP')                                                      │                │
./iterative_design_nand.log:│ │                path = PosixPath('${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod')                                                                                                                  │                │
./iterative_design_nand.log:│   12                                                                                          │ path = PosixPath('${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod') │                                                  │
./iterative_design_nand.log:│ ❱ 1027 │   │   with self.open(mode='r', encoding=encoding, errors=errors) as f:                 │     self = PosixPath('${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod') │                                            │
./iterative_design_nand.log:│   1016 │   │   """                                                                              │      self = PosixPath('${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod') │                                           │
./iterative_design_nand.log:FileNotFoundError: [Errno 2] No such file or directory: '${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod'
(.venv) faebryk-wt1 % grep -R 'DIP-14_W7.62mm.kicad_mod' .
./iterative_design_nand.log:│ │                path = PosixPath('${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod')                                                                                                                  │                │
./iterative_design_nand.log:│   12                                                                                          │ path = PosixPath('${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod') │                                                  │
./iterative_design_nand.log:│ ❱ 1027 │   │   with self.open(mode='r', encoding=encoding, errors=errors) as f:                 │     self = PosixPath('${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod') │                                            │
./iterative_design_nand.log:│   1016 │   │   """                                                                              │      self = PosixPath('${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod') │                                           │
./iterative_design_nand.log:FileNotFoundError: [Errno 2] No such file or directory: '${KICAD7_FOOTPRINT_DIR}/Package_DIP.pretty/DIP-14_W7.62mm.kicad_mod'
(.venv) faebryk-wt1 % 
```

### Expected Behaviour

The program exited with all checks "ok"

### Possible Solution

#pr number

### Version

df1e405

### Relevant log output

```shell
See attached files above
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

iopapamanoglou, 2024-11-03:
[image]
```python
 global_fp_lib_table = C_kicad_fp_lib_table_file.loads(GLOBAL_FP_LIB_PATH)
 ```

```python
    case "darwin":
        GLOBAL_FP_LIB_PATH = (
            Path("~/Library/Preferences/kicad").expanduser()
            / KICAD_VERSION
            / "fp-lib-table"
        )
```

Seems like your global fp-lib-table wasn't converted to kicad8 yet

[image]

faebryk expects 'KICAD8_FOOTPRINT_DIR'

```python
        dir_path = Path(
            lib.uri.replace("${KICAD8_FOOTPRINT_DIR}", str(GLOBAL_FP_DIR_PATH))
        )
```

iopapamanoglou, 2024-11-03:
Can you try to migrate and see if that fixes it?
We could add something to make the fail more gracious if this is indeed the reason it fails.

mawildoer, 2024-11-03:
For some ungodly reason my global `fp-lib-table` still points to KiCAD 7 stuff and wouldn't migrate.
[image]

Did it manually with a find-replace and this solved it!

Thanks mate!!!

closed 2024-11-03 by mawildoer